rmk-rs/rmk #304: can't get Aurora Sweep with nice!nano_v2 working

patmuk, 2025-03-30:
I have a [Autota Sweep from splitkb]((https://docs.splitkb.com/product-guides/aurora-series/schematics/aurora-sweep) with a [nice!nano_v2]((https://nicekeyboards.com/docs/nice-nano/pinout-schematic).

I tried both, the [keyboard.toml based configuration](https://github.com/patmuk/rmk_pat/blob/main/keyboard.toml) and the [rust based configuration](https://github.com/patmuk/rmk_pat/tree/rust_config) - these links point to my public repository.

Both have the same behavior, but I can't figure out why and how I can proceed:

The project compiles, flashing (with Softdevices S140 6.1.1) works.
When unplugging my Mac (OS 15.3.2, M3 Max) recognizes the bluetooth keyboard - but no button presses are recognized.
When I klick on "identify this keyboard" I am prompted to press the key right of shift - but, again, no button press is recognized.

I don't have any hardware probes. I do have displays connected to the n!ns.

I don't really understand what are the input and the output pins.My board uses a matrix, I set the row pins to the input pins and the column pins to output.

I read somewhere on the manufacturers page (splitkb.com) that a key is active on "low" - but I can't find that anymore. However, setting "direct_pin_low_active = true" makes no difference and seems to be available for direct pins only?

Any help or pointers would be highly appreciated!

HaoboGu, 2025-03-30:
according to the schematic, your keyboard is row2col, the default setting of RMK is col2row. You can try the workaround here: https://haobogu.github.io/rmk/faq.html#my-matrix-is-row2col-the-matrix-doesnt-work

patmuk, 2025-03-30:
Many thanks! This worked! However, I think my configuration is still messed up. (Only the [rust based configuration](https://github.com/patmuk/rmk_pat/tree/rust_config) works, but that is ok) I changed to row2col, as you suggested, and corrected the pins. The left side works, but the keys are different than in my keymap.rs. On the peripheral side some keys work, but when I press certain ones the keyboard crashes.

Can you please explain the meaning of the input and output pins? Or point me to an explanation (did not find any)?
I guess I would need to understand the schemata better: Is the input pin going into the source of the diode? 

Then, in `config_matrix_pins_nrf!`, how the pins are to be ordered? I ordered them from
```
input: col0, col1, col2, col3, col4
out: row0, row1, row2, row3
```
and not in the order the pins are physically on the MCU.

In `run_rmk_split_central` I have values for the row/col offset (which I set to 0) but in `run_rmk_split_peripheral`there is no offset value.

==> in a split configuration, is `central` the right side? I am used from ZMK that the left side is the main one ...

I did not mention it: Many thanks for this great project! I didn't use QMK, because I wanted a wireless kb, and ZMK recently broke for me because of some upstream update. Nevertheless, the turnaround cycle with the server-side only build was too high anyways.
Your project looks perfect!!!

HaoboGu, 2025-03-31:
QMK has a great doc which explains this: https://docs.qmk.fm/how_a_matrix_works

In one word, the direction of the matrix could always be like: 

```
output_pin =>   >|   => input_pin
                 ↑
              diode(be aware of it's direction)
```

So if the matrix is row2col, the output pins are rows, and input pins are cols

> In run_rmk_split_central I have values for the row/col offset (which I set to 0) but in run_rmk_split_peripheral there is no offset value.

Yes, for the split side, the `run_rmk_split_peripheral` only have the number of rows/cols of this split peripheral. Only central knows all offsets. [here](https://haobogu.github.io/rmk/split_keyboard.html#central) is the documentation. The `run_peripheral_manager` runs on central and it manages the split. If you have the wrong config, say `row_num + row_offset > total number of rows`, the keyboard would panic because a non-existing position is triggered.

HaoboGu, 2025-03-31:
From the aurora sweep's schematics, the setting should be like:

Central: 
let input_pins = [B6, D4, C6, D7, E6]
let output_pins = [F6, F5, F7, B1]

Peripheral:
let input_pins = [B5, B4, E6, D7, C6]
let output_pins = [B1, F7, F6, B3]

Those pins should be converted to nice!nano's configs, I don't know how those pins are mapped to nice!nanos pin.

HaoboGu, 2025-03-31:
If the keys are still messed up, you can post your vial.json here. vial.json should be consist from the matrix config.

patmuk, 2025-03-31:
Ah, many thanks for the great explanation! I overlooked the offset configuration for the peripheral. Got the pins correct now, all keys work and the kb doesn't panic :) The letters are not in the place I thought they should be - but I will figure this out. 
I don't yet understand the relation of the vial.json and the keymap.rs. I followed the vial.json documentation and created a physical layout of my board - but this resulted in a board with more rows than I actually have - e.g. top row has 2 keys (instead of 5), next row 3 (instead of being in the same row) ...

Nevertheless, I will figure it out on my own now to not steal more of your precious time :) Will write here in case I have further problems.

I hope you don't mind if I improve the documentation a tiny bit, adding the information you gave me here. I will do a PR for review :)

HaoboGu, 2025-03-31:
vial.json defines the **keymap's layout** while keymap.rs defines the actually keys on the position. In vial.json, you have a `(row, col)` on each position, you should have a key defined on that position in keymap.rs, aka `keymap[row][col]`. Yeah, I have to admit it's a little bit complex to understand all those definitions, keymap, layout, matrix, etc. I had a hard time as well 😄

> I hope you don't mind if I improve the documentation a tiny bit, adding the information you gave me here. I will do a PR for review :)

Sure! I'd be very happy if you want to help to improve the documentations! :D

patmuk, 2025-03-31:
Ah, super, thanks for the explanation!

patmuk, 2025-03-31:
I fixed all mistakes and got everything working, thanks again!

However, in the Vial.app the right-most thumb key on the left side is not shown. The key and key mapping from the rust configuration works, but the key is simply invisible in vial.
In my [vial.json](https://github.com/patmuk/rmk_pat/blob/rust_config/vial.json) I specified only 4 keys in the last row, while the keymap.rs defines 10 (the 6 not existing ones of my aurora sweep (Ferris Sweep like) as well. However, adding these keys in the   vial.json didn't help - 9 keys were shown, but the missing key was still missing.

Nevertheless, this is not a problem for me, as I plan to configure the mapping via rust and use vial only for visualizing/bugfixing.
Probably it is a bug in the vial.app - but I can't see what is so special about the missing key (which I labeled "3.4").

HaoboGu, 2025-04-01:
sure ；）

closed 2025-04-01 by patmuk