rmk-rs/rmk #315: Bug: th! not working in lt!

patmuk, 2025-04-03:
When I enter a layer with lt! (holding the key), a th! configured team doesn't output anything (neither the tap key, nor the hold key).

patmuk, 2025-04-04:
Actually no KeyAction (tap-hold, layer-tap-hold, ...) works within another KeyAction.

Another example is: I have a shifted letter (`!`: shift!(Kc1) ) and want to assign a HRM on it ( mt!(shift!(1), LSFT). Without macros (as we can't nest them) this would be:
```
    const LSFT: ModifierCombination = ModifierCombination::from_bits(0b00010);
    let exclamation_mark = KeyAction::WithModifier(Action::Key(KeyCode::Kc1), LSFT);
    KeyAction::ModifierTapHold(exclamation_mark, LSFT)
```
Doesn't work, since the first argument of `ModifierTapHold` needs to be an `Action`, not a `KeyAction`.

How can we solve this?
Tl;dr: With Macros? (#308): It would need to be a Action::Macro(Vec<Action>).

HaoboGu, 2025-04-04:
Yes, I tried to avoid recursion there. I'm not sure whether it's worth to introduce recursion, maybe we can have some other way to achieve it, but I totally have no idea now..

patmuk, 2025-04-04:
Yes, this was a good decision :) I think that macros would help here again :) Meaning, for my example, something like `Action::Macro([Action::Modifier(LSFT), Action::Key(KeyCode::Kc1)])` should work and would still avoid recursion :)

HaoboGu, 2025-08-21:
Closing it because it seems to be already solved by #308

closed 2025-08-21 by HaoboGu