rmk-rs/rmk #317: bug: KeyCode::RepeatKey maps to th!(Again, Meh)

patmuk, 2025-04-08:
When configuring the keycap with k!(RepeatKey) it results wrongly in th!(Again, Meh) (as can be seen in via.

I assume the byte encoding goes wrong, as `RepeatKey = 0x779` and `Again = 0x079`. Probably combining `Meh` + `Again` results in `0x779` as well.

HaoboGu, 2025-04-08:
Yes you are correct. 0x779 is th!(Again, Meh) in vial keycode. 

[image]

vial keycode for repeat key is 0x7C79

patmuk, 2025-04-08:
Indeed :) But where is this call missing?

HaoboGu, 2025-04-08:
https://github.com/HaoboGu/rmk/blob/596feb4935ff07b527baff787c49c02a302d02b9/rmk/src/via/keycode_convert.rs#L12-L24

Now if the key is Repeat(0x779), `to_via_keycode` does nothing on it, resulting in via keycode 0x779 aka `th!(Again, Meh)`. So adding conversion from `0x779` to `0x7C79` is enough:

```rust
 KeyAction::Single(a) => match a { 
     Action::Key(k) => { 
         if k.is_macro() { 
             k as u16 & 0xFF | 0x7700 
         } else if k.is_user() { 
             k as u16 & 0xF | 0x7E00 
         } else if k.is_combo() || k.is_boot() { 
             // is_rmk() 's subset 
             k as u16 & 0xFF | 0x7C00 
         } else if k as u16 == 0x779 {
             0x7C79
         } else { 
             k as u16 
         } 
     } 
```

patmuk, 2025-04-09:
Doing so I discovered a deeper bug:
Basically
`KeyCode::from_primitive(1913)`
leads to `KeyCode::No`

(although `KeyCode::from_primitive(0x7779)` converts correctly to `KeyCode::RepeatKey`.

I assume that the num_enum crate doesn't handle gaps in the numeration in the `enum KeyCode`.

Nevertheless, hardcoding as you proposed would work ... tough there might be other keys which are mapped wrongly.

But I wonder why the code is different at all? The other keys seem to align with [QMK's KeyCodes](https://github.com/qmk/qmk_firmware/blob/a0f40d9410d1ecac4b498b999ebd04e65ef762f4/quantum/keycodes.h#L4), which in turn are the same as [the linux hid implementation](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-input.c).

The `RepeatKey` is not a standard key - why not to simply give it the same code as in Vial and QMK (0x7C79)?

HaoboGu, 2025-04-10:
I've merged a fix #324 for repeat key.

As for the keycode compatibility, it's a historical problem.. When I was implementing it, I used enums for representation of key actions. In QMK/Vial it's just a u16. The `KeyAction`s are different, for example, in RMK you can have tap-hold for more keys, so keeping all of them same is impossible, that's why the keycodes are different.

But now I feel that it can be improved, by using vial's keycode everywhere and adding a mapping between vial's keycode and Rust enum representation. For key actions that cannot be represented by vial's keycode, just use an un-occupied u16. It requires a full refactoring of the Key representation system in RMK, I don't think it has high priority now.

HaoboGu, 2025-04-11:
#324 was merged, closing

closed 2025-04-11 by HaoboGu