rmk-rs/rmk #319: Introduce a (optional?) matrix transformer layer

tib888, 2025-04-08:
The introduction of a matrix transformer layer would make the congig file easier to read/write, when the "electronic" key matrix does not match the phisical reality.
(Also, after this, the key positions will be identified with only a singlesequence number instead of two coordinates.)

My example use case:
```
# My full "electronic" matrix looks like this (this could be implicitly generated by default):
columns = 12
rows = 5 
mx = """(0, 5) (0, 4) (0, 3) (0, 2) (0, 1) (0, 0)      (0, 6) (0, 7) (0, 8) (0, 9) (0, 10) (0, 11)
        (1, 5) (1, 4) (1, 3) (1, 2) (1, 1) (1, 0)      (1, 6) (1, 7) (1, 8) (1, 9) (1, 10) (1, 11)
        (2, 5) (2, 4) (2, 3) (2, 2) (2, 1) (2, 0)      (2, 6) (2, 7) (2, 8) (2, 9) (2, 10) (2, 11)
        (3, 5) (3, 4) (3, 3) (3, 2) (3, 1) (3, 0)      (3, 6) (3, 7) (3, 8) (3, 9) (3, 10) (3, 11)
        (4, 5) (4, 4) (4, 3) (4, 2) (4, 1) (4, 0)      (4, 6) (4, 7) (4, 8) (4, 9) (4, 10) (4, 11)"""
        
        
# I have populated only these switches:
mx = """ 
                      (1, 3) (1, 2) (1, 1) (1, 0)      (1, 6) (1, 7) (1, 8) (1, 9) 
        (2, 5) (2, 4) (2, 3) (2, 2) (2, 1) (2, 0)      (2, 6) (2, 7) (2, 8) (2, 9) (2, 10) (2, 11)
        (3, 5) (3, 4) (3, 3) (3, 2) (3, 1) (3, 0)      (3, 6) (3, 7) (3, 8) (3, 9) (3, 10) (3, 11)
        (4, 5) (4, 4)          (4, 3) (4, 2) (4, 1)  (4, 7) (4, 8) (4, 9)          (4, 10) (4, 11)"""

# So currently I have to describe my layers with a row jump at the sides:
    [
        ["No", "No", "No", "No", "No", "No",                                    "No", "No", "No", "No", "No", "No"],
        ["No", "No", "Y", "K", "R", "B",                                        "F", "C", "O", "U", "No", "No"],
        ["LCtrl",  "G", "MT(S, LAlt)", "MT(T,LShift)", "MT(L, LCtrl)", "V",     "J", "MT(N, RCtrl)", "MT(A, RShift)", "MT(E, LAlt)", "Semicolon", "Insert"],
        ["LShift", "MT(M, LGui)", "P", "W", "H", "Q",                           "X", "Z", "Dot", "Comma", "MT(I, RGui)", "Quote"],
        ["LAlt", "D", "LT(4, Escape)", "LT(2, Space)", "LT(5, Tab)", "No",      "No", "LT(5, Enter)", "LT(2, Backspace)", "LT(4, Delete)", "Slash", "LGui"]
    ]
    
# If there would be (an optional?) matrix transformer like this:
mxmap = """(2, 5) (2, 4) (1, 3) (1, 2) (1, 1) (1, 0)      (1, 6) (1, 7) (1, 8) (1, 9) (2, 10) (2, 11)
           (3, 5) (3, 4) (2, 3) (2, 2) (2, 1) (2, 0)      (2, 6) (2, 7) (2, 8) (2, 9) (3, 10) (3, 11)
           (4, 5) (4, 4) (3, 3) (3, 2) (3, 1) (3, 0)      (3, 6) (3, 7) (3, 8) (3, 9) (4, 10) (4, 11)
                                  (4, 2) (4, 1) (4, 0)  (4, 6) (4, 7) (4, 8)                         """        

# Then all key codes would end up in a simple sequence, which matches the reality:    
["LCtrl",  "G",           "Y",           "K",             "R",            "B",                                             "F",                "C",            "O",             "U",           "Semicolon",   "Insert",
 "LShift", "MT(M, LGui)", "MT(S, LAlt)", "MT(T,LShift)",  "MT(L, LCtrl)", "V",                                             "J",                "MT(N, RCtrl)", "MT(A, RShift)", "MT(E, LAlt)", "MT(I, RGui)", "Quote",
 "LAlt",   "D",           "P",           "W",             "H",            "Q",                                             "X",                "Z",            "Dot",           "Comma",       "Slash",       "LGui",
                                                          "LT(4, Escape)", "LT(2, Space)", "LT(5, Tab)",   "LT(5, Enter)", "LT(2, Backspace)", "LT(4, Delete)"]

# This could be described with a single string, which would result a much nicer config:
"""
 LCtrl  G           Y           K             R            B                               F                C            O             U           Semicolon   Insert
 LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V                               J                MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
 LAlt   D           P           W             H            Q                               X                Z            Dot           Comma       Slash       LGui
                                      LT(4, Escape) LT(2, Space) LT(5, Tab)   LT(5, Enter) LT(2, Backspace) LT(4, Delete)]
"""    
```

HaoboGu, 2025-04-08:
In this situation, keyboard.toml should contain 3 parts:

1. mx
2. mxmap
3. keymap

Am I right?

Is there any easy conversion from qmk/zmk keymap configuration?

tib888, 2025-04-08:
> In this situation, keyboard.toml should contain 3 parts:
> 
> 1. mx
> 2. mxmap
> 3. keymap
> 
> Am I right?

No, I believe, mx can be omitted, I included it only to make the example easier to understand.

The full matrix dimensions can be computed the usual way: the split parts stick together to make a big matrix.
Only 'mxmap' is needed to define a sequence, which will be used in the later configuration steps.

The keymap could contain only one string per layer.

ZMK uses such transformation already in their .dtsi file: (in this example the colums of left half are "mirrored"):
```
default_transform: default_transform {
        // See https://zmk.dev/docs/development/hardware-integration/new-shield#matrix-transform
        // for how to configure this
        compatible = "zmk,matrix-transform";
        columns = <12>;
        rows = <5>;

        map = <
            RC(0, 5) RC(0, 4) RC(0, 3) RC(0, 2) RC(0, 1) RC(0, 0)      RC(0, 6) RC(0, 7) RC(0, 8) RC(0, 9) RC(0, 10) RC(0, 11)
            RC(1, 5) RC(1, 4) RC(1, 3) RC(1, 2) RC(1, 1) RC(1, 0)      RC(1, 6) RC(1, 7) RC(1, 8) RC(1, 9) RC(1, 10) RC(1, 11)
            RC(2, 5) RC(2, 4) RC(2, 3) RC(2, 2) RC(2, 1) RC(2, 0)      RC(2, 6) RC(2, 7) RC(2, 8) RC(2, 9) RC(2, 10) RC(2, 11)
            RC(3, 5) RC(3, 4) RC(3, 3) RC(3, 2) RC(3, 1) RC(3, 0)      RC(3, 6) RC(3, 7) RC(3, 8) RC(3, 9) RC(3, 10) RC(3, 11)
            RC(4, 5) RC(4, 4) RC(4, 3)   RC(4, 2) RC(4, 1) RC(4, 0)  RC(4, 6) RC(4, 7) RC(4, 8)   RC(4, 9) RC(4, 10) RC(4, 11)
        >;
    };
```

Then the keymap looks like this:
```
    keymap {
        compatible = "zmk,keymap";

        default_layer {
            display-name = "Base QWERTY";
            bindings = <
&none           &none         &none         &kp LBKT        &kp RBKT       &kp BSLH         &kp N9   &kp N0         &kp MINUS       &none         &none            &none
&kp NUHS        &kp Q         &kp W         &kp E           &kp R          &kp T            &kp Y    &kp U          &kp I           &kp O         &kp P            &kp EQUAL
&kp GRAVE       &lhrm LGUI A  &lhrm LALT S  &lhrm LSHIFT D  &lhrm LCTRL F  &kp G            &kp H    &rhrm RCTRL J  &rhrm RSHIFT K  &rhrm LALT L  &rhrm RGUI SEMI  &kp SQT
&none           &kp Z         &kp X         &kp C           &kp V          &kp B            &kp N    &kp M          &kp COMMA       &kp DOT       &kp FSLH         &none
&studio_unlock  &none         &none         &kp ESC         &kp SPACE      &lt L_NAV TAB    &kp RET  &kp BSPC       &kp DEL         &none         &hun_aaAA_       &studio_unlock
            >;
        };
```

A bigger question is: is it acceptable that the current RMK configs have to be converted to a new format?

HaoboGu, 2025-04-08:
> A bigger question is: is it acceptable that the current RMK configs have to be converted to a new format?

If it eliminates the layout & matrix config in keyboard.toml and vial.json, provides more intuitive way to config the keyboard's matrix/layout/keymap, I think it's worth to do it.

tib888, 2025-04-08:
Kanata also starts with such a map (the context is a bit different since it is running on PC not on the keyboard):
```
(defsrc
`	1	2	3	4	5	6	7	8	9	0	-	=
q	w	e	r	t	y	u	i	o	p	[	]	\
a	s	d	f	j	k	l	;	'
ContextMenu	ArrowLeft	ArrowRight	ArrowUp	ArrowDown
)
```
Then the configuration looks like this:
```
(deflayer U_BASE
`	S-1	S-2	S-3	S-4	S-5	S-6	S-7	S-8	S-9	S-0	-	=
q	w	e	r	t	y	u	i	o	p	[	]	@\-mod
@a-mod	@s-mod	@d-mod	@f-mod	@j-mod	@k-mod	@l-mod	@;-mod	'
@context-mod ArrowLeft ArrowRight	ArrowUp	ArrowDown
)
```
Which is again, pretty similar.

Unfortunately I have not tried QMK yet.

tib888, 2025-04-08:
> > A bigger question is: is it acceptable that the current RMK configs have to be converted to a new format?
> 
> If it eliminates the layout & matrix config in keyboard.toml and vial.json, provides more intuitive way to config the keyboard's matrix/layout/keymap, I think it's worth to do it.

vial.json contains the "graphical" coordinates for keys in Vial, so unfortunately we can not get rid of it.
But if mxmap is present in the keyboard.toml, then it would be enough to write the sequence number on the keys...

tib888, 2025-04-08:
To interpret the keymap strings, pest' parser crate could be used, which would give more flexibility in the future and for better error messages...

HaoboGu, 2025-04-08:
It looks like to put the row/col labels in vial.json to mxmap, introduce a default keymap representation which is consistent with the mxmap?

I am thinking can we do it better, for example putting the KLE json（with or without vial labels) in the keyboard.toml and removing vial.json? The current way adds the transformation but does not simplify them

tib888, 2025-04-09:
If mx-map would be written in the keyboard toml, the simplification would be that in KLE it would be enough to write the sequence number on the keys, no col,row would be needed there.

If you have some strange matrix to save some pins, then this transformation also helps a lot. 
*(which I do: in an other "standard" full size keyboard for example the whole numpad is mapped into the 7th row...)

Also getting rid of this noise:
[
["","","",""],
["","","",""],
["","","",""],
],
And using instead only a multiline string is nice too:
"""

"""

I would like to also introduce aliases:
```
[aliases]
m = "MT(M, LGui)"
s = "MT(S, LAlt)"
t = "MT(T,LShift)"
l = "MT(L, LCtrl)" 
...
```

so the long lines in this:
```
"""
 LCtrl  G           Y           K             R            B                               F                C            O             U           Semicolon   Insert
 LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V                               J                MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
 LAlt   D           P           W             H            Q                               X                Z            Dot           Comma       Slash       LGui
"""  
```

could be written much more compact:
```
"""
 LCtrl  G  Y  K  R  B    F  C  O  U  Semicolon   Insert
LShift @m @s @t @l V    J  @n @a @e @i          Quote
LAlt   D  P  W     H    Q  X  Z  Dot Comma Slash LGui
"""  
```

tib888, 2025-04-09:
On the other hand, there are other keyboard layout editors, which look much nicer than vial, like https://github.com/nickcoutsos/keymap-editor
And besides this, ZMK started its own development.
So the question is: how long do we want to stick with Vial? Is there a plan or started effort for an RMK UI? 

nickcoutsos or example has a bit different .json format:
(It describes the matrix col, row, the sequence number and the xy coordinate, rotation of keys.)
```
{
  "layouts": {
    "LAYOUT": {
      "layout": [
	{ "row": 0, "col":  0, "x":     0, "y":  0 },
        { "row": 0, "col":  1, "x":     1, "y":  0 },
        { "row": 0, "col":  2, "x":     2, "y":  0 },
        { "row": 0, "col":  3, "x":     3, "y": -0.5 },
        { "row": 0, "col":  4, "x":     4, "y":  0 },
        { "row": 0, "col":  5, "x":     5, "y":  0 },
        { "row": 0, "col":  7, "x":     9, "y":  0 },
        { "row": 0, "col":  8, "x":    10, "y":  0 },
        { "row": 0, "col":  9, "x":    11, "y": -0.5 },
        { "row": 0, "col": 10, "x":    12, "y":  0 },
        { "row": 0, "col": 11, "x":    13, "y":  0 },
        { "row": 0, "col": 12, "x":    14, "y":  0 },
		
        { "row": 1, "col":  0, "x":     0, "y":  1 },
        { "row": 1, "col":  1, "x":     1, "y":  1 },
        { "row": 1, "col":  2, "x":     2, "y":  1 },
        { "row": 1, "col":  3, "x":     3, "y":  0.5 },
        { "row": 1, "col":  4, "x":     4, "y":  1 },
        { "row": 1, "col":  5, "x":     5, "y":  1 },
        { "row": 1, "col":  7, "x":     9, "y":  1 },
        { "row": 1, "col":  8, "x":    10, "y":  1 },
        { "row": 1, "col":  9, "x":    11, "y":  0.5 },
        { "row": 1, "col": 10, "x":    12, "y":  1 },
        { "row": 1, "col": 11, "x":    13, "y":  1 },
        { "row": 1, "col": 12, "x":    14, "y":  1 },
    
        { "row": 2, "col":  0, "x":     0, "y":  2 },
        { "row": 2, "col":  1, "x":     1, "y":  2 },
        { "row": 2, "col":  2, "x":     2, "y":  2 },
        { "row": 2, "col":  3, "x":     3, "y":  1.5 },
        { "row": 2, "col":  4, "x":     4, "y":  2 },
        { "row": 2, "col":  5, "x":     5, "y":  2 },
        { "row": 2, "col":  7, "x":     9, "y":  2 },
        { "row": 2, "col":  8, "x":    10, "y":  2 },
        { "row": 2, "col":  9, "x":    11, "y":  1.5 },
        { "row": 2, "col": 10, "x":    12, "y":  2 },
        { "row": 2, "col": 11, "x":    13, "y":  2 },
        { "row": 2, "col": 12, "x":    14, "y":  2 },
    
        { "row": 3, "col":  0, "x":     0, "y": 3 },
        { "row": 3, "col":  1, "x":     1, "y": 3 },
        { "row": 3, "col":  2, "x":     2, "y": 3 },
        { "row": 3, "col":  3, "x":     3, "y": 2.5 },
        { "row": 3, "col":  4, "x":     4, "y": 3 },
        { "row": 3, "col":  5, "x":     5, "y": 3 },
        { "row": 3, "col":  7, "x":     9, "y": 3 },
        { "row": 3, "col":  8, "x":    10, "y": 3 },
        { "row": 3, "col":  9, "x":    11, "y": 2.5 },
        { "row": 3, "col": 10, "x":    12, "y": 3 },
        { "row": 3, "col": 11, "x":    13, "y": 3 },
        { "row": 3, "col": 12, "x":    14, "y": 3 },
    
	 { "row": 4, "col":  0, "x":     0, "y": 4 },
        { "row": 4, "col":  1, "x":     1, "y": 4 },
        { "row": 4, "col":  2, "x":     2, "y": 4 },
        { "row": 4, "col":  3, "x":   3.5, "y": 4.5, "r": -24, "rx": 4, "ry": 4 },
        { "row": 4, "col":  4, "x":     5, "y": 4.2},
        { "row": 4, "col":  5, "x":   6.5, "y": 4.5, "r":  24, "rx": 7, "ry": 4 },
        { "row": 4, "col":  7, "x":   7.5, "y": 4.5, "r": -24, "rx": 8, "ry": 4 },
        { "row": 4, "col":  8, "x":     9, "y": 4.2},
        { "row": 4, "col":  9, "x":  10.5, "y": 4.5, "r":  24, "rx": 11,"ry": 4 }
        { "row": 4, "col": 10, "x":    12, "y": 4 },
        { "row": 4, "col": 11, "x":    13, "y": 4 },
        { "row": 4, "col": 12, "x":    14, "y": 4 }
      ]
    }
  }
}
```

Anyway including these jsons in keyboard.toml is possible.

HaoboGu, 2025-04-09:
I'm OK with the alias and single string keymap representation. I'm thinking about simplify the physical electronic representation, too. 

And conversion from qmk/zmk config to the new representation is also important, at least for the matrix and layout. 

> Is there a plan or started effort for an RMK UI?

Yes, definitely. But it's in very early stage, I don't think it could be finished soon.

tib888, 2025-04-09:
So, should I start working on this?

HaoboGu, 2025-04-09:
I prefer to start with the multiline string representation and matrix transformation, what do you think?

HaoboGu, 2025-04-09:
How to represent layers? Add a layer identifier as what kanata does like:
```
"""
layer1:
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
layer2:
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
"""  
```

tib888, 2025-04-09:
I proposed to use one string per layer:
```
layers = [
"""
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
""",
"""
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
"""  
]
```

Or if we want to be able to name the layers:
```
[layers]
default = """
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
"""
upper = """
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
"""
```
then these nemas could be used instead of simple numbers.
For example: TG(#upper)

HaoboGu, 2025-04-09:
What do you think to use toml's [Array of Tables](https://toml.io/en/v1.0.0#array-of-tables) like:

```
[[layer]]
# name is optional
name = "default"
keys = """
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
"""
[[layer]]
name = "layer2"
keys = """
LCtrl  G           Y           K             R            B            F            C            O            U            Semicolon   Insert
LShift MT(M, LGui) MT(S, LAlt) MT(T,LShift)  MT(L, LCtrl) V            J            MT(N, RCtrl) MT(A, RShift) MT(E, LAlt) MT(I, RGui) Quote
LAlt   D           P           W             H            Q            X            Z            Dot           Comma       Slash       LGui
"""
```

The layer order conforms to the layer config order in the toml

> If missing, the assumption would be a simpe grid matrix with row number by col number dimensions.

Then the vial.json with labelled positions is still needed? It's very rare to have keys on every position, vial should know the mapping between key layout and the matrix position :/

tib888, 2025-04-09:
Array of Tables syntax is fine - that way more layer specific properties can be added later.

tib888, 2025-04-09:
I guess a transition period will be needed, when both the old and this new configuration method works.
Do you foresee any conflicts?

HaoboGu, 2025-04-09:
I don't know. Maybe we should write documentation and a number of example configs to see if there is any conflicts?

tib888, 2025-04-09:
What about the rust code based configs, will they get this maping transformation also and change to store the key bindings in a one dimensional array instead of two?

tib888, 2025-04-09:
I have updated the keyboard config documentation as "specification".

https://github.com/tib888/rmk/blob/feat/matrix_map/docs/src/keyboard_configuration.md

Please review/comment!

HaoboGu, 2025-04-10:
It looks really great! Intuitive and easy to understand(at least for me hahaha).

A few comments from me:

> The matrix_map is a string built from (row, col) coordinate pairs, listed in the same order as you want to define your keys in your key map

If I understand it correctly, (row, col) is the position in the electronic matrix. I think we should have a simple introduction about what `row = 0` represents, help users understand the position in matrix better

> For example, if you set a keycode Backspace, it will be turned to KeyCode::Backspace. So you have to ensure that the keycode string is valid, or RMK wouldn't compile!

I've added many [alias](https://github.com/HaoboGu/rmk/blob/main/rmk-macro/src/keycode_alias.rs) can be used here and changed the key representation case-insensitive, but I forgot to update the documentation here LOL. I'd be very appreciated if you can help me update docs here.

> MyCut = "WM(x, LCtrl)"

I prefer `my_cut`

As for the Rust code config, I prefer to not change anything.

HaoboGu, 2025-04-10:
> As for the Rust code config, I prefer to not change anything.

In this case, if users use Rust code, a full labelled vial.json is still needed..

With the toml config way, the users could just use a "raw" vial.json, and the stored compressed vial.json in the keyboard is generated by combining toml config and "raw" vial.json.

tib888, 2025-04-10:
I have added the requested changes. Should I open a PR for easier commenting/collaboration"?

tib888, 2025-04-10:
> In this case, if users use Rust code, a full labelled vial.json is still needed..
>
> With the toml config way, the users could just use a "raw" vial.json, and the stored compressed vial.json in the [...]
I can not fully follow this, but during the implementation everything will find its place...

HaoboGu, 2025-04-10:
> I have added the requested changes. Should I open a PR for easier commenting/collaboration"?

Sure, please do

HaoboGu, 2025-04-10:
> Also, shouldn't we use one based indexes for matrix coordinates to be more intuitive for non programmers?

Both are fine for me

tib888, 2025-04-11:
https://github.com/HaoboGu/rmk/pull/327 is ready for testing.
For now it does only the minimal processing required and fills up the LayoutConfig as usual.
Mixed usage with the old config style is allowed, but [[layer]] items will start form the 0th index.

Here is the relevant part of my config, how ended up (these aliases are VERY useful!):
```toml

[aliases]
"~"="WM(Grave, LShift)"
"`"="Grave"
"!"="WM(Kc1, LShift)"
"@"="WM(Kc2, LShift)"
"#"="WM(Kc3, LShift)"
"$"="WM(Kc4, LShift)"
"%"="WM(Kc5, LShift)"
"^"="WM(Kc6, LShift)"
"&"="WM(Kc7, LShift)"
"*"="WM(Kc8, LShift)"
"("="WM(Kc9, LShift)"
")"="WM(Kc0, LShift)"
"-"="Minus"
"_"="WM(Minus, LShift)"
"="="Equal"
"+"="WM(Equal, LShift)"
"["="LeftBracket"
"{"="WM(LeftBracket, LShift)"
"]"="RightBracket"
"}"="WM(RightBracket, LShift)"
"\\"="Backslash"
"|"="WM(Backslash, LShift)"
";"="Semicolon"
":"="WM(Semicolon, LShift)"
"'"="WM(Quote, LShift)"
"\""="WM(Quote, LShift)"
","="Comma"
"<"="WM(Comma, LShift)"
"."="Dot"
">"="WM(Dot, LShift)"
"/"="Slash"
"?"="WM(Slash, LShift)"
cut="WM(X, LCtrl)"
copy="WM(C, LCtrl)"
paste="WM(V, LCtrl)"
paste_noformat="WM(V, LCtrl|LShift)"
find="WM(F, LCtrl)"
replace="WM(H, LCtrl)"
undo="WM(Z, LCtrl)"
redo="WM(Y, LCtrl)"
m="LT(navigation, M)"
s="MT(S, LAlt)"
t="MT(T, LShift)"
l="MT(L, LCtrl)"
n="MT(N, RCtrl)"
a="MT(A, RShift)"
e="MT(E, LAlt)"
i="LT(symbols, I)"

[layout]
rows = 4
cols = 12
layers = 10
matrix_map = """
(1, 0) (1, 1) (0, 2) (0, 3) (0, 4) (0, 5)              (0, 6) (0, 7) (0, 8) (0, 9) (1, 10) (1, 11)
(2, 0) (2, 1) (1, 2) (1, 3) (1, 4) (1, 5)              (1, 6) (1, 7) (1, 8) (1, 9) (2, 10) (2, 11)
(3, 0) (3, 1) (2, 2) (2, 3) (2, 4) (2, 5)              (2, 6) (2, 7) (2, 8) (2, 9) (3, 10) (3, 11)
                           (3, 2) (3, 3) (3, 4)  (3, 7) (3, 8) (3, 9)
""" 

[[layer]]
name = "default" #0 (Hunglish)
keys ="""
LCtrl   G  Y  K  R  B                  F  C  O  U @; Insert
LShift @m @s @t @l  V                  J @n @a @e @i Quote
LAlt    D  P  W  H  Q                  X  Z @, @= @. LGui
LT(functions, Escape) LT(symbols, Space) LT(navigation, Tab)  LT(symbols, Enter) LT(navigation, Backspace) LT(functions, Delete) 
"""

[[layer]]
name = "symbols" #1 
keys ="""
@~ @@  @{ @} @! @|    @! Kp7 Kp8 Kp9 KpSlash KpAsterisk
@` @\\ @( @) @" @&    @: Kp4 Kp5 Kp6 KpMinus KpPlus
@$ @#  @[ @] @' @^    @? Kp1 Kp2 Kp3 Equal   @%
@< @_ @>              @; Kp0 KpDot 
"""

```

HaoboGu, 2025-04-11:
Did you see this: https://github.com/HaoboGu/rmk/pull/310 ? 

I think we can have those alias built-in and write a doc for it. So that everyone can use the simplified alias in the keymap

tib888, 2025-04-11:
The aliases are all about personalization, so I would not fill the aliases table with anything by default.
For example on hungarian kayboard layout totally differents keycombinatios are neded to produce the same symbols.
If one needs that table, just copy paste it in his config.

HaoboGu, 2025-04-11:
ah yeah, I mean the commonly used part, such as `,` for comma, `.` for dot, `=` for equal, etc

tib888, 2025-04-11:
Unfotunately, even that is different! 

[image]

tib888, 2025-04-13:
Lets clarify auto-generation of vial.json, using this as example:

```toml
matrix_map = """
(1, 0) (1, 1) (0, 2) (0, 3) (0, 4) (0, 5)              (0, 6) (0, 7) (0, 8) (0, 9) (1, 10) (1, 11)
(2, 0) (2, 1) (1, 2) (1, 3) (1, 4) (1, 5)              (1, 6) (1, 7) (1, 8) (1, 9) (2, 10) (2, 11)
(3, 0) (3, 1) (2, 2) (2, 3) (2, 4) (2, 5)              (2, 6) (2, 7) (2, 8) (2, 9) (3, 10) (3, 11)
                           (3, 2) (3, 3) (3, 4)  (3, 7) (3, 8) (3, 9)
"""
```

This is read into:
```rust
sequence_to_grid: Vec<(u8, u8)> = [
(1, 0),(1, 1),(0, 2),(0, 3),(0, 4),(0, 5),(0, 6),(0, 7),(0, 8),(0, 9),(1, 10),(1, 11),
(2, 0),(2, 1),(1, 2),(1, 3),(1, 4),(1, 5),(1, 6),(1, 7),(1, 8),(1, 9),(2, 10),(2, 11),
(3, 0),(3, 1),(2, 2),(2, 3),(2, 4),(2, 5),(2, 6),(2, 7),(2, 8),(2, 9),(3, 10),(3, 11),
(3, 2),(3, 3),(3, 4),(3, 7),(3, 8),(3, 9)
];
```

and at the same time we build the reverse map:
(we know that the not 'None' switches are polulated from the full "electrinic grid")
```
grid_to_sequence: Vec<Vec<Option<u32>>> = [
[None   None   2,     3,     4,     5,     6,     7,     8,     9,     None,   None],
[0,     1,     14,    15,    16,    17,    18,    19,    20,    21,    10,     11],
[12,    13,    26,    27,    28,    29,    30,    31,    32,    33,    22,     23],
[24,    25,    36,    37,    38,    None,  None,  39,    40,    41,    34,     35],
];
```

I guess, that this json should be also built for vial (assumming that the '\n' in the matrix_map config string means a new row on the keyboard, and we need center aligned keys):
```json
[
["1,0","1,1","0,2","0,3","0,4","0,5","0,6","0,7","0,8","0,9","1,10","1,11"],
["2,0","2,1","1,2","1,3","1,4","1,5","1,6","1,7","1,8","1,9","2,10","2,11"],
["3,0","3,1","2,2","2,3","2,4","2,5","2,6","2,7","2,8","2,9","3,10","3,11"],
[{x:3},"3,2","3,3","3,4","3,7","3,8","3,9"]
]
```
So, in KLE this would appear like this (so Vial will show the same):

[image]

So naturally we will NOT be able to present the real layout, just this approximation.
The real in this case looks like this:

[image]

Do you agree, or have better ideas?

tib888, 2025-04-15:
I was thinking that you want to achieve that the new users should be able to use RMK without using KLE at all (with the help of an  auto-generated vial.json).

Once the user needs to use KLE, its not much difference if he has to label the keys by (row, col) coordinates or with simple sequence numbers, like on the last image. But of course, sequence numbers are more user friendly, and we have the info in matrix_map to convert the sequence numbers to coordinates, if we make matrix_map obligatory.

HaoboGu, 2025-04-15:
> I was thinking that you want to achieve that the new users should be able to use RMK without using KLE at all (with the help of an auto-generated vial.json).

I did think that, but if we're using vial, we still need similar representation.. and generate the KLE json str as a result. I cannot propose a better yet KLE(vial)-compatible representation. That's why I changed my mind. 🥲

tib888, 2025-04-15:
Then maybe in the documentation we could give hint, for example:

Assuming this is your matrix_map:
```
(1, 0) (1, 1) (0, 2) (0, 3) (0, 4) (0, 5)              (0, 6) (0, 7) (0, 8) (0, 9) (1, 10) (1, 11)
(2, 0) (2, 1) (1, 2) (1, 3) (1, 4) (1, 5)              (1, 6) (1, 7) (1, 8) (1, 9) (2, 10) (2, 11)
(3, 0) (3, 1) (2, 2) (2, 3) (2, 4) (2, 5)              (2, 6) (2, 7) (2, 8) (2, 9) (3, 10) (3, 11)
                           (3, 2) (3, 3) (3, 4)  (3, 7) (3, 8) (3, 9)
````
if you replace '(' to '",'  and ')' to '",' and add the [...], brackets around each line like this:
```
["1,0","1,1","0,2","0,3","0,4","0,5","0,6","0,7","0,8","0,9","1,10","1,11"],
["2,0","2,1","1,2","1,3","1,4","1,5","1,6","1,7","1,8","1,9","2,10","2,11"],
["3,0","3,1","2,2","2,3","2,4","2,5","2,6","2,7","2,8","2,9","3,10","3,11"],
["3,2","3,3","3,4","3,7","3,8","3,9"]
```
then this string can be copied into the "Raw data" tab of KLE. 
And after that only the positions have to be nicely adjusted to match the physical layout, and download the final json.

Of course the numpad example could be also used...

tib888, 2025-04-15:
Maybe this raw_vial.json could be a side-product of the build.

But this whole thing is not important to me. We can simply close this issue too.

HaoboGu, 2025-04-15:
Fine. Let's leave it just as is now and see if there's better solution in the future

closed 2025-04-15 by tib888