rmk-rs/rmk #338: [NRF52840] Board does not auto switch to USB mode if you plugin without 3.3V

howardng97, 2025-04-14:
🔌 USB-C Port Does Not Work Immediately Without External Power
When using the Pro Micro nRF52840 board, note the following behavior:
The firmware does not initialize USB immediately when connecting the board directly via the USB-C port.
As a result, keyboard input (or other USB-based communication) will not work — you won’t be able to type any characters.
✅ Workaround / Solution
Power the board with 3.3V first, then plug in the USB-C cable:
Use an external source like a J-Link, ST-Link, or other debug adapter to provide 3.3V to VDD
After the board is powered, connect the USB-C cable
The firmware will now properly initialize USB, and keyboard input will work

HaoboGu, 2025-04-15:
ahh, high voltage mode is related to the hardware design, which means power is connected to directly to the vddh(no LDO).

howardng97, 2025-04-15:
I'm using the SuperMini NRF52840 from [this list of alternatives](https://github.com/joric/nrfmicro/wiki/ALternatives). However, I've found it quite tricky to get the device working properly.

The board came preloaded with the Adafruit UF2 bootloader and SoftDevice S140 v6.1.1, which allows for easy drag-and-drop flashing via UF2. However, when I tried to upgrade the SoftDevice to S140 v7.3.0 by flashing a UF2 file, and flash the firmware build with cargo to UF2

To recover it, I had to manually reconnect the SWD (Serial Wire Debug) lines to a J-Link debugger, and then re-flash the correct S140 v7.3.0 SoftDevice directly using the Nordic tools. After that, I used probe-rs with cargo run to flash my firmware, and it finally worked.

HaoboGu, 2025-04-15:
By default RMK uses SD 6.1.1. If you updated SD to 7.3.0, you should change memory.x to match the SD version too.

As for the not working issue, may I ask how do you compile your firmware? could you try the latest main branch? (which doesn't need SD anymore)

HaoboGu, 2025-04-15:
it seems that promicro52840 uses high voltage mode. As a fix you can add 

https://github.com/HaoboGu/rmk/blob/713eeb642351694948859db4237d713edb199b01/examples/use_rust/nrf52840_ble/src/main.rs#L103

to the MCU initializer.

howardng97, 2025-04-16:
Ok I'll try

dbldippy, 2025-05-15:
@howardng97 I see the same issue with current main. I'm using a config example. Did the fix above work for you?

howardng97, 2025-05-15:
not yet, I need to turn my  battery on, so I can use the keyboard

closed 2025-08-21 by HaoboGu