rmk-rs/rmk #352: No action on "LShift + Kc8" and "LShift + Kc9" combinations, Dell Vostro 1310 laptop keyboard

vladimiry, 2025-04-26:
The goal was to convert a laptop keyboard (keyboard matrix), specifically Dell Vostro 1310 as a first prototype, into a desktop keyboard.

Steps taken:
- Have wired the keyboard FPC lines into the arbitrary MCU GPIO pins. 
- Decoded the keyboard matrix with the https://github.com/vladimiry-playground/keyboard-firmware-dell-vostro-1310-shared/blob/master/data/decoded-keyboard-matrix.png result. So, according to the table, the LShift key connects the PC6xPB14 pins.
- Have put that matrix decoding table into https://github.com/vladimiry-playground/keyboard-firmware-dell-vostro-1310-shared/blob/master/src/keymap.rs and the table headers into https://github.com/vladimiry-playground/keyboard-firmware-dell-vostro-1310-shared/blob/90aa5e1720adf6ec500402990884c91aab0a1988/src/main.rs#L78-L79 as an input/output arrays.
- The Storage and Vial things not used in the project.

The prototype is ready and everything seems to work fine except for the named keys combinations - simply nothing happens. It works fine with the RShift though, and the Kc8/Kc9 keys being pressed individually work well too. Also, generally LShift works well in combination with keys, but not with Kc8/Kc9. Anything I could do to tackle the issue?

I tried to swap LShift/RShift keys in the keymap config - the issue persists at the same LShift/PC6xPB14 position.
I didn't try to re-wire PC6/PB14 pins into some other ones available on the MCU board, but it looks like I will have to.
RMK is the first keyboard firmware I've ever used, so maybe I'm doing something wrong.

vladimiry, 2025-04-26:
Just noticed that some other combinations don't work with LShift, but run fine with RShift. Like for example, Ctrl+Shift+O - opening bookmarks manager in the browser. So it's not just LShift+Kc8 and LShift + Kc9 combinations.

vladimiry, 2025-04-26:
It turned out to be weirder than I thought. Just realized that "LCtrl+~" also NOOP, but "RCtrl+~" is ok. The issue cause remains unclear at the moment, and I have to admit that this keyboard unfortunately is unusable until the issue gets resolved.

HaoboGu, 2025-04-27:
Yes, it's quite weird. If the `LShift` works but some combination don't work, I suspect it's a hardware issue, for example, it's not a regular key matrix with diodes. Do you have schematics of your board?

vladimiry, 2025-04-27:
> Do you have schematics of your board?

Nope. But that should be a regular laptop keyboard matrix. I thought any laptop kb matrix supposed to work, but now I see that it's not that straightforward. Anyway, I have got few more matrixes from other laptops, maybe I get luck with them.

vladimiry, 2025-04-27:
Closing as this is unlikely to be a firmware issue, but hardware.

closed 2025-04-27 by vladimiry

vladimiry, 2025-04-27:
I got it working in QMK, so the hardware part is verified as valid. But still, could not yet make it work with RMK flawlessly on the same hardware it worked using QMK. Maybe one day I try it on another keyboard. 

With QMK, there was no original issue (NOOP keys combinations), but another one. The 11 keys were NOOP. I looked into the keymap (matrix decoding table), and realized that it's the full top row was NOOP. Here it is:

[image]

Clearly, the PB2 pin is responsible for the top row here. So I re-wired the PB2 line to a free pin PD10 hoping that it's some pin/hardware related issue, and changed the `matrix_pins` field of QMK's `keyboard.json` config accordingly, and it worked out.

The nice thing about QMK is that getting out of suspend works there, as I noted in #353, which is an essential feature for me. 

What I don't like about QMK, aside it's legacy nature, it's verbosity and relative complexity (this was the second kb firmware I tried, after RMK), is that it forces me to fill `layouts` field in `keyboard.json` I don't get why I have to, as, if I get it right, the above-displayed matrix decoding table should be sufficient to make the keyboard work, and it's already defined in `keymap.c` plus `matrix_pins` field of `keyboard.json`. So I had to please QMK by filling `layouts` with some data by simply putting there a semi-stub data as if my keyboard was `COL*ROW` chess board. Obviously, in reality the keys are not equally sized and rows have a different number of keys, etc, but putting "chess board" there was sufficient to please QMK. Not very pleasant dev experience, but the outcome is ok.

HaoboGu, 2025-04-28:
@vladimiry Thanks for the feedback! After you changed the pin, do you still have the problem in RMK? 

Do you have `async_matrix` feature enabled?

vladimiry, 2025-04-29:
> After you changed the pin, do you still have the problem in RMK?

Of course, I did right after got QMK working, there was no difference.

> Do you have async_matrix feature enabled?

I got: `input: [PD10, PC6, PB0, PA4, PA0, PC4, PA2, PA6]`. So channels/banks 6/0/4 being used several times. But I didn't find an easy way to define an array with `ExtiInput::new(p.<PIN>, p.EXTI<PIN_CH>, Pull::Down)` where some channels/banks, like 6/0/4, being re-used as Rust borrow checking thing doesn't allow such scenario (`value moved here`...). I guess I could re-wire the input pins at the hardware side in order to use unique `p.EXTI<PIN_CH>` channels/banks, but I'm not ready yet to do the re-wiring for this project, but I should be able to take this into the consideration in a next keyboard project.

HaoboGu, 2025-04-29:
oh you're using async matrix? for stm32 I suggest to use normal input/output pin and don't use Exti. There's limitations for Exit in stm32

HaoboGu, 2025-04-29:
yes there is limitations when using Exti on stm32. I suggest to disable async_matrix feature.