rmk-rs/rmk #353: Geeting out of suspend by key pressing doesn't work

vladimiry, 2025-04-26:
I used to get the laptop out from suspend by pressing any key on an external keyboard, but it doesn't work here. I press the keys, but the laptop remains suspended.
The project is here https://github.com/vladimiry-playground/keyboard-firmware-dell-vostro-1310-shared.

vladimiry, 2025-04-27:
I've just tested it using QMK with the same keyboard matrix as named in #352 and the same MCU setup, and it works there.

Not sure, but maybe the issue is specific to my test keyboard, as it's not yet well suited to being managed by RMK firmware, as there are still some issues listed in #352. So maybe someone else could share the getting of suspend experience using RMK firmware.

HaoboGu, 2025-05-06:
@vladimiry Which OS are you using?

HaoboGu, 2025-05-06:
Oh, I guess [this issue](https://github.com/HaoboGu/rmk/issues/188) is still here. It seems a Linux related bug.

vladimiry, 2025-05-06:
It works if QMK firmware used and with any other regular keyboard, so it doesn't look like a system/Linux bug. Anyway, let's close this issue as a duplicate of #188.

closed 2025-05-06 by vladimiry

HaoboGu, 2025-05-06:
yeah, I mean it's a RMK bug related to Linux😄. I'll investigate it later!