rmk-rs/rmk #355: Can the peripheral use the same ble_addr() approach as the central?

IniterWorker, 2025-04-28:
Of course, this is not released yet, but I was wondering if the peripheral could also use the same address generation approach as the central?

Specifically, I am referring to the following function:

```rust
fn ble_addr() -> [u8; 6] {
    let ficr = embassy_nrf::pac::FICR;
    let high = u64::from(ficr.deviceid(1).read());
    let addr = high << 32 | u64::from(ficr.deviceid(0).read());
    let addr = addr | 0x0000_c000_0000_0000;
    unwrap!(addr.to_le_bytes()[..6].try_into())
}
```

Reference: [peripheral.rs#L115-L115](https://github.com/HaoboGu/rmk/blob/ab9a28bcb887deeae7b7de81a4125b5a75811541/examples/use_rust/nrf52840_ble_split/src/peripheral.rs#L115C5-L115C64)

Thanks!

HaoboGu, 2025-04-28:
Of course, it can!

IniterWorker, 2025-04-28:
Let me know if you need PRs for it.
Thank you for your quick response!

It's working:
https://github.com/IniterWorker/rmk-lily58-keyboard/commit/6f6fbdbd8186ebe69aad5eb0e1229ba19f8f8e86

But you will need to clear the storage on the peripheral.

HaoboGu, 2025-04-28:
> Let me know if you need PRs for it.

yes, thanks!

> But you will need to clear the storage on the peripheral.

oh, that's correct. `clear_storage` should be setting. Good catch!

PR #359 (2025-05-04): Use ble_addr in nrf split peripheral clear_storage
Close #355
closed 2025-05-04 by HaoboGu