rmk-rs/rmk #357: NRF52840 BLE not advertised, blue LED blinking

dbldippy, 2025-05-01:
First off, thank you @HaoboGu for helping to bring Rust to the keeb community. Great work. Love to get involved to bring Miryoku crbkd to RMK

I am having difficulties connecting to a nice!nano keeb, a Corne in my case. I went by the fairly detailed instructions, but they don't seem to be working for me

What I work with:

- Wireless Corne with nice!nano V2, NRF52840 S140 6.1.1 
- MacBook Pro Nov. 2024

What I've done so far:

- use rmkit to create a NRF52840 split BLE config-based project
- made sure that the linker script matches the uc version
- did not (need to) modify the project
- built successfully  
- via USB, flashed the central uf2 to the left half, the peripheral uf2 to the right half
- disconnect the USB cable completely
- let my MBP forget my daily driver BLE corne, just to be sure 
 
What I observe:
 
- the RMK keeb is not showing up in the macOS bluetooth settings
- the blue programmable LED of the nice!nano is blinking. This indicates that it is in bootloader mode, I think

Am I missing something obvious? Thanks for any help

HaoboGu, 2025-05-01:
Your operation looks correct. Could you post your `memory.x` or your project here? I can have a look

dbldippy, 2025-05-01:
It's unchanged from the generated project:

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  /* FLASH : ORIGIN = 0x00000000, LENGTH = 1024K */
  /* RAM : ORIGIN = 0x20000000, LENGTH = 256K */

  /* These values correspond to the NRF52840 with Softdevices S140 6.1.1 */
  FLASH : ORIGIN = 0x00026000, LENGTH = 824K

  /* These values correspond to the NRF52840 with Softdevices S140 7.3.0 */
  /* FLASH : ORIGIN = 0x00027000, LENGTH = 820K */
  RAM : ORIGIN = 0x20020000, LENGTH = 128K
}
```

HaoboGu, 2025-05-02:
looks correct. Could you try to connect USB again to see whether the bootloader driver appears again? if that, then I suspect the uf2 is not correct created or flashed. 

You can try the following command to re-generate the firmware:

```
cargo clean
cargo make uf2 --release
```

If you still have the issue, you can post your project link here and I could have a try

dbldippy, 2025-05-04:
[Here](https://github.com/dbldippy/corne-rs) is the project. I created it with `rmkit init // select 'split' and 'NRF52840'`

A project created with `rmkit init // select 'standard' and 'NRF52840'` connects with bluetooth. Just the split doesn't

HaoboGu, 2025-05-04:
Your project lacks `memory.x`. I added it manually:

```
MEMORY
{
  /* These values correspond to the NRF52840 with Softdevices S140 6.1.1 */
  FLASH : ORIGIN = 0x00026000, LENGTH = 824K

  RAM : ORIGIN = 0x20020000, LENGTH = 128K
}
```

And compiled the firmware using `cargo make uf2 --release`.

Then I flashed the generated `corne-rs-central.uf2` to the board, it works correctly.

I guess there's something wrong on your board? Could you try to clear the chip totally and reflash the bootloader?

dbldippy, 2025-05-05:
I re-flashed the bootloader S140 6.1.1 including softdevice according to [these instrlctions](https://nicekeyboards.com/docs/nice-nano/troubleshooting/#my-nicenano-seems-to-be-acting-up-and-i-want-to-re-flash-the-bootloader)

I updated the project with the `memory.x` file, then compiled just like you do. The sha256 of the `corne-rs-central.uf2` is
`c1e0ee95c06e98fe7d654d7c624ac5749ecc31b9c536009947d8d41b471a3926`

After flashing via USB, the problem persists. Tried the other keeb half. Same thing

Other env info:

rmkit V0.0.11
rustc 1.86.0

```
objdump -h target/thumbv7em-none-eabihf/release/central

target/thumbv7em-none-eabihf/release/central:	file format elf32-littlearm

Sections:
Idx Name            Size     VMA      LMA      Type
  0                 00000000 00000000 00000000
  1 .vector_table   00000100 00026000 00026000 DATA
  2 .text           000203f8 00026100 00026100 TEXT
  3 .rodata         00004d0c 000464f8 000464f8 DATA
  4 .data           00000054 20035df0 0004b208 DATA
  5 .gnu.sgstubs    00000000 0004b260 0004b260 DATA
  6 .bss            00009db8 20035e44 20035e44 BSS
  7 .uninit         00000400 2003fbfc 2003fbfc BSS
  8 .defmt          0000016b 00000000 00000000
  9 .debug_loc      0004095e 00000000 00000000 DEBUG
 10 .debug_abbrev   00001b7e 00000000 00000000 DEBUG
 11 .debug_info     000e8629 00000000 00000000 DEBUG
 12 .debug_aranges  00004370 00000000 00000000 DEBUG
 13 .debug_ranges   0001e2a0 00000000 00000000 DEBUG
 14 .debug_str      0015ce2d 00000000 00000000 DEBUG
 15 .comment        00000099 00000000 00000000
 16 .ARM.attributes 00000038 00000000 00000000
 17 .debug_frame    00005020 00000000 00000000 DEBUG
 18 .debug_line     00029079 00000000 00000000 DEBUG
 19 .debug_pubnames 000002be 00000000 00000000 DEBUG
 20 .debug_pubtypes 00000047 00000000 00000000 DEBUG
 21 .symtab         0000a990 00000000 00000000
 22 .shstrtab       000000fd 00000000 00000000
 23 .strtab         000213d1 00000000 00000000
```

I'm not sure if its the bootloader. When I flash any of the halves with the uf2 of a rmkit-generated standard project, they are advertised and they can connect

HaoboGu, 2025-05-05:
> Also, when I flash any of the halves with the uf2 of a rmkit-generated standard project, they are advertised and they connect
> 
> Also updated the project with the memory.x file, then compiled just like you do. After flashing via USB, the problem persists. Tried the other keeb half. Same thing

what's the difference between the working try and the not working try? Only memory.x?

dbldippy, 2025-05-05:
Afaics, the only difference between the working and the non-working firmware is that one is a split, the other a normal non-split. I created both projects with rmkit and left them unchanged. The `memory.x` is the same for both projects

I reset the settings on the two halves with the ZMK reset firmware and flashed central again. Same behavior. Blue light blinking, one reset push sets the board into bootloader mode

How do I do the full reset you mentioned?

HaoboGu, 2025-05-05:
The full reset requires you having a debug probe. Run `probe-rs erase --chip nRF52840` to fully erase the chip.

I checked the config file in the unibody & split template again, the only difference is the used pin. Could you try changing the pin?

dbldippy, 2025-05-06:
You were right. I adjusted the pins according to [this config](https://github.com/Nicolai-Holmgaard/rmk-config/blob/81670fdd15df05105e142a82bce0067deddab0b5/keyboard.toml). Now, the keeb boots correctly. Thanks for your help!!

HaoboGu, 2025-05-06:
Good to hear that. But it's still kind of strange.. I checked the pins, didn't see any issues.. Could you please share your hardware schematics of your board? I want to ensure that whether it's related to the hardware.

HaoboGu, 2025-05-06:
Okay, it seems that nice!nano uses P0_14 as the reset pin, which is also used in the template.. I'll update the template, use other pins available on nice!nano. Thanks for the feedback!

closed 2025-05-06 by HaoboGu