rmk-rs/rmk #362: LTs are getting stuck

IniterWorker, 2025-05-06:
I have 4 LTs that I can use on a regular basis. Sometimes the LTs get stuck. When I think the planets are aligned and I get multiple LTs at the same time, the issue occurs. I didn’t run a debug session, but the problem happened twice in a row, and I had to reset the board each time.

* Platform: nrf 
* Ref: ab9a28bcb887deeae7b7de81a4125b5a75811541

Keyboard project: https://github.com/IniterWorker/rmk-lily58-keyboard

HaoboGu, 2025-05-06:
Yeah, there's some tap-hold bugs. @hitsmaxft is working on it.

hitsmaxft, 2025-05-06:
> the

enable hrm in behaviour config will fix the stuck , but multi LT still buggy , and i'm working on it

HaoboGu, 2025-07-04:
Should be fixed by #371, @IniterWorker Could you have a try latest version?

IniterWorker, 2025-07-08:
@HaoboGu  I'm not sure if this behavior is intentional or a bug, but the current handling of the event feels quite nice. A quick double press seems to bypass the TAP_HOLD, which actually improves the experience. I've tried to reproduce the same conditions, and overall it feels much better now. The tap-hold action is smoother and doesn't block as much as it used to.

HaoboGu, 2025-07-08:
> A quick double press seems to bypass the TAP_HOLD, which actually improves the experience.

Yes, it looks like `prior_idle_time` are working. Glad to hear that the new version works for you :D

HaoboGu, 2025-07-15:
I'm closing this issue as it's solved by the last version.