rmk-rs/rmk #363: LT being triggered before timeout on simultaneous key presses

eduardobraun, 2025-05-06:
When typing "fast" I'm getting unwanted LT triggers, my current hrm config is
```
tap_hold = { enable_hrm = true, prior_idle_time = "120ms", hold_timeout = "400ms", post_wait_time = "0ms" }
```
I tried removing the post wait and increasing the hold timeout, but it seems to have no effect on this,
if I tap `H` which is a `RAlt_t(H)` and the normal key `E` really fast, I`m getting something like
```
êheheheheêheêheheheêêêheheêheheheheheheêêheêêêheheêêêêheêêheêheêêêêêê
```
`ê` is the result of `RAlt + e` from my xkb layout, but it should only happen when I hold `H` not on taps, I think the problem is when `e` is released before `h` it triggers the LT even if the timeout did not expire.

HaoboGu, 2025-05-06:
Could you try to increase `prior_idle_time`? This is also #276 wants to solve.

eduardobraun, 2025-05-06:
Sure, I'll try it. But I think my description was lacking in details,
if I type `he` fast without delays it happens only for the first character:
`êheheehehheheheheheheheheheheheeheh`
if I wait in between each `he` then I get a result like in the previous message.

eduardobraun, 2025-05-06:
I believe I've found the issue.. 

https://github.com/HaoboGu/rmk/blob/rmk-v0.6.1/rmk/src/keyboard.rs#L745-L756

If another key is pressed the `hold_timeout` is not considered, it keeps looping and treat it as a hold.
I made this change:
```
                        // self.process_key_action_normal(hold_action, key_event).await;
                        self.process_key_action_tap(tap_action, key_event).await;
```
and it works as I expect it would, not sure if this breaks something else.. it means a press in another key before the `hold_timeout` will cancel the hold and threat as tap.
If this is the expected behavior, probably its better to exit before the loop though..

HaoboGu, 2025-05-07:
what do you expect for the `[hold][press][release][timeout][unhold]` sequence? In the default setting, RMK behaves like [permissive hold](https://docs.qmk.fm/tap_hold#tap-or-hold-decision-modes) in QMK: the hold action will triggered for the key press & release even it happens before the timeout

eduardobraun, 2025-05-07:
In this case I expect it to trigger the hold fuction and apply its modifier to the key.

Looking at the QMK docs I'm having trouble with `Nested tap (ABBA)`, the default behavior results in `ab` and the permissive hold results in `B`.

Permissive hold makes home rows mods unusable for me, I'm not precise enough with my typing to avoid this case, it seems to happen all the time and I keep triggering keys in other layers.. To type in this mode I must press keys one by one making sure I release them before the next key, that is too slow.

This is the first layer of the layout I'm trying to use:
```
// Left
[ "KC_TRNS", "KC_Q", "KC_W", "KC_E", "KC_R", "LT3(KC_T)" ],
[ "KC_TRNS", "KC_A", "LGUI_T(KC_S)", "LCTL_T(KC_D)", "LSFT_T(KC_F)", "RALT_T(KC_G)" ],
[ "KC_TRNS", "KC_Z", "KC_X", "KC_C", "LT2(KC_V)", "LT1(KC_B)" ],
[ -1, -1, -1, "KC_TAB", "KC_SPACE", "KC_ESCAPE" ],
// Right
[ "KC_Y", "KC_U", "KC_I", "KC_O", "KC_P", "KC_GRAVE" ],
[ "RALT_T(KC_H)", "RSFT_T(KC_J)", "LCTL_T(KC_K)", "LGUI_T(KC_L)", "KC_SCOLON", "KC_QUOTE" ],
[ "KC_N", "KC_M", "KC_COMMA", "KC_DOT", "KC_SLASH", "KC_BSLASH" ],
[ "KC_BSPACE", "KC_ENTER", "KC_TRNS", -1, -1, -1 ]
```
I expected that it would be safe to press the keys as long as I released them before `hold_timeout`.

HaoboGu, 2025-05-07:
I got it. Let's see how can it be improved, maybe after #276

eduardobraun, 2025-07-13:
Its really good now, I didn't test it extensively, but I tried to trigger the hrm by pressing the keys just as before and it works as expected! Thank you.

HaoboGu, 2025-07-15:
I'm closing this issue as it's solved by the last version.

closed 2025-07-15 by HaoboGu