rmk-rs/rmk #364: Dynamic master

turtle-bazon, 2025-05-09:
Interested in next feature - keyboard with 3 parts - one central and two parts. When they are together then central part is master and two peripherals is peripherals as usual. But when this parts far from master and master is not reachable then one of peripheral take master role. I didn't found this in documentation. Is this implemented?

HaoboGu, 2025-05-09:
no. that means you have to connect two different Bluetooth devices to your PC.

This could also be confusing when the dynamic split couldn't connect to the master but the last peripheral can. You will find two keyboards but each of them only have a half working.

It's technically feasible, but requires lots lots of work and tuning.

HaoboGu, 2026-08-29:
I don't see a solution possible in the near future, closing this right now. But I'm still open for any ideas of dynamic central support!