rmk-rs/rmk #368: Corne: halves are not pairing

dbldippy, 2025-05-15:
Hello @HaoboGu. Congrats on the replacement of SoftDevice with trouble. However, I have to report that I'm not able anymore to use both halves of my keeb. Central works fine, but peripheral does not pair

As prescribed [here](https://zmk.dev/docs/troubleshooting/connection-issues), I flash both halves with the reset firmware, then flash both halves with freshly built nrf52840_ble_split example

Also, when central is paired with my Mac, and I push the corne's reset, it does not advertise anymore

I also noticed the 3V power issue [#338](https://github.com/HaoboGu/rmk/issues/338)

I can imagine that replacing the BT stack can have all kinds of effects in the beginning. So, I'm just reporting what I see. Appreciate your work

HaoboGu, 2025-05-15:
Oh this is a known issue. There is a workaround that you can try:

1. Set `clear_storage` to true for both peripheral & central
2. Flash both central & peripheral firmware
3. Set `clear_storage` back to false and compile
4. Flash both splits again

hitsmaxft, 2025-05-18:
> Oh this is a known issue. There is a workaround that you can try:
> 
> 1. Set `clear_storage` to true for both peripheral & central
> 2. Flash both central & peripheral firmware
> 3. Set `clear_storage` back to false and compile
> 4. Flash both splits again
> 

I suggest providing an improvement point: when validation fails, disable storage, and let user can resetting storage through vial.

HaoboGu, 2025-05-20:
But vial cannot reset peripherals.. I'm still seeking for better solutions