rmk-rs/rmk #401: Q: Babblepaste

FoxPride, 2025-06-07:
Hi!
I would like to know if it is possible to replicate something similar to [Babblepaste](https://github.com/milestogo/qmk_firmware/blob/m2gkeymap/users/miles2go/babblePaste.md) with current code?

In short, this is a macro that allows you to achieve stable behavior for operations in different OS (copy, paste, etc. on one button).

The idea is that we set some variable that represents the current OS. When we press the macro to copy, we check the variable and call the corresponding combo.

FoxPride, 2025-06-07:
What if set OS variable and then change it with some macro instead of detection? For example, with Bluetooth profile change.

HaoboGu, 2025-06-07:
I don't think it will solve the problem because you don't know what is the *current value* of that "OS variable" without having a display on the keyboards.

HaoboGu, 2025-06-07:
Or you have to create a keycode for every support platform, such as "SetWindows", "SetMacos", etc. But it's quite verbose and I don't like it.

tib888, 2025-06-21:
The easiest way to implement this would be to use [forks](https://rmk.rs/docs/features/configuration/behavior.html#fork)
The detected (or set) OS should be encoded in some flags, and aollow the forks to act on thise flags too, not just on modifiers.

FoxPride, 2025-06-22:
My idea was something similar - just assume that we starting at certain OS and then switch to the one you need right now.

Regarding original question: I was wandering if I can replicate similar functionality with current rmk features.