rmk-rs/rmk #436: sticking homerow mods especially in conjunction with combos

willpuckett, 2025-07-04:
I had time to do some testing this week around #278. (working repository https://github.com/willpuckett/sessile/tree/main/rmk)

There have been a lot of awesome changes to RMK and I'm not sure I made all the updates correctly, in particular the changes to memory.x (I'm using a xiao ble with the seeed bootloader), but things seem to be compiling and running somewhat. 

I'm experiencing some murky and indeterminate problems surrounding home row mods.

when I run the board with full home row mods and combos utilizing them, the board runs for a minute then starts to devolve into **stuck mods** and **repeating keys**—even when I'm not touching it.

when I change the home row keys to normal letters (no MT's), the board works largely as expected.

I thought this was just a problem with hrm's, but when I switched the combos to use the normal keys (no MT's) and then reactivated the hrms, the problem improved considerably.

So, to test things further, I completely disabled all the combos while enabling the home row mods, and the board runs without crashing/repeating keys etc.

I also enabled hrms and then activated only combos which do not involve a home row key. The board is much more stable and generally works, but hrm's do get stuck after a while....

This issue may be similar to #363 or #362, but it seems to be arising in conjunction with combos and MT. 

Thanks again for all the great work!

willpuckett, 2025-07-04:
basically, to duplicate, include a MT key in a combo

willpuckett, 2025-07-04:
Home row mods are also do not seem to be modding combos

HaoboGu, 2025-07-05:
Thanks a lot for the feedback! I'll look into it soon

HaoboGu, 2025-07-05:
@willpuckett Could you give #440 a try?

HaoboGu, 2025-07-05:
> Also, since this is the first time I'm posting something here, let me thank you [@HaoboGu](https://github.com/HaoboGu), and the other maintainers for this amazing firmware! I've been eyeing the releases slowly since September and am excited that the firmware is starting to be a viable alternative to ZMK.

Thanks for the post! I'm really happy to hear that #440 solves the problem :D

willpuckett, 2025-07-05:
Yes just testing now it seems to resolve all the issues. TH mods are working on combos, TH mods are not getting stuck, and I can type normally (rolly) without the whole thing jamming... 

Thanks for the quick fix #440 just made RMK my daily driver!

closed 2025-07-05 by willpuckett