rmk-rs/rmk #441: Single keystroke results in a repeated character

jerkydev, 2025-07-05:
First of all, since it's my first message here let me say thank you for your amazing work!

I've tried RMK recently for the first time. Everything seems to work, however, I noticed annoying character repeats (sometimes it's just single repeat and sometimes even double).

I can't reproduce the same issue with the same keyboard and QMK firmware. I have also tried replacing the mechanical switches, however this has had no effect in RMK.

I tried to ketch something with the logging through USB, however, found there just this on character repeat:

```
Process key action after combo processing: Single(Key(Dot)), KeyEvent { row: 2, col: 8, pressed: true }
[TAP-HOLD] tap_hold_decision: permissive=true, is_tap_hold_buffered=false, is_pressed=true, action=Single(Key(Dot))
[TAP-HOLD] --> decision is : Ignore
Process Single key action: Key(Dot), KeyEvent { row: 2, col: 8, pressed: true }
Last key code changed from  Dot to Dot(pressed: true)
Process key action after combo processing: Single(Key(Dot)), KeyEvent { row: 2, col: 8, pressed: false }
[TAP-HOLD] tap_hold_decision: permissive=true, is_tap_hold_buffered=false, is_pressed=false, action=Single(Key(Dot))
[TAP-HOLD] --> decision is : Ignore
Process Single key action: Key(Dot), KeyEvent { row: 2, col: 8, pressed: false }
Record released key event: KeyEvent { row: 2, col: 8, pressed: false }
Process key action after combo processing: Single(Key(Dot)), KeyEvent { row: 2, col: 8, pressed: true }
[TAP-HOLD] tap_hold_decision: permissive=true, is_tap_hold_buffered=false, is_pressed=true, action=Single(Key(Dot))
[TAP-HOLD] --> decision is : Ignore
Process Single key action: Key(Dot), KeyEvent { row: 2, col: 8, pressed: true }
Last key code changed from  Dot to Dot(pressed: true)
Process key action after combo processing: Single(Key(Dot)), KeyEvent { row: 2, col: 8, pressed: false }
[TAP-HOLD] tap_hold_decision: permissive=true, is_tap_hold_buffered=false, is_pressed=false, action=Single(Key(Dot))
[TAP-HOLD] --> decision is : Ignore
Process Single key action: Key(Dot), KeyEvent { row: 2, col: 8, pressed: false }
Record released key event: KeyEvent { row: 2, col: 8, pressed: false }
```

I'm using RMK version `0.7.4` and here is my `keyboard.toml`

```
[keyboard]
name = "piantor-rmk"
product_name = "piantor-rmk"
vendor_id = 0xbeeb
product_id = 0x0002
manufacturer = "beekeeb"
chip = "rp2040"

[layout]
rows = 4
cols = 10
layers = 4
keymap = [
    [
        ["Q", "W", "E", "R", "T", "Y", "U", "I", "O", "P"],
        ["MT(A, LCtrl)", "MT(S, LAlt)", "MT(D, LShift)", "MT(F, LGui)", "G", "H", "MT(J, LGui)", "MT(K, LShift)", "MT(L, LAlt)", "MT(Semicolon, LCtrl)"],
        ["Z", "X", "C", "V", "B", "N", "M", "Comma", "Dot", "Slash"],
        ["No", "No", "_", "Space", "MO(1)", "MO(2)", "MO(3)", "_", "No", "No"]
    ],
    [
        ["WM(1, LShift)", "WM(2, LShift)", "WM(3, LShift)", "WM(4, LShift)", "WM(5, LShift)", "_", "_", "_", "_", "_"],
        ["1", "2", "3", "4", "5", "_", "_", "_", "_", "_"],
        ["F1", "F2", "F3", "F4", "F5", "_", "_", "_", "_", "_"],
        ["No", "No", "_", "_", "_", "_", "_", "_", "No", "No"]
    ],
    [
        ["_", "_", "_", "_", "_", "WM(6, LShift)", "WM(7, LShift)", "WM(8, LShift)", "WM(9, LShift)", "WM(0, LShift)"],
        ["_", "_", "_", "_", "_", "6", "7", "8", "9", "0"],
        ["_", "_", "_", "_", "_", "F6", "F7", "F8", "F9", "F10"],
        ["No", "No", "_", "_", "_", "_", "_", "_", "No", "No"]
    ],
    [
        ["MediaPrevTrack", "MediaPlayPause", "MediaNextTrack", "AudioVolDown", "AudioVolUp", "Escape", "CapsLock", "PageUp", "Backspace", "Delete"],
        ["_", "_", "_", "_", "_", "Left", "Down", "Up", "Right", "Enter"],
        ["Bootloader", "_", "_", "BrightnessDown", "BrightnessUp", "Home", "Tab", "PageDown", "End", "_"],
        ["No", "No", "_", "_", "_", "_", "_", "_", "No", "No"]
    ],
]

[behavior]
tap_hold = { enable_hrm = true, prior_idle_time = "120ms", hold_timeout = "250ms", post_wait_time = "50ms", permissive_hold = true, chordal_hold = true}

[storage]
enabled = false

[dependency]
defmt_log = false

[split]
connection = "serial"

[split.central]
rows = 4
cols = 5
row_offset = 0
col_offset = 0
serial = [{ instance = "PIO0", tx_pin = "PIN_1", rx_pin = "PIN_1" }]
[split.central.matrix]
matrix_type = "normal"
input_pins = ["PIN_7", "PIN_8", "PIN_9", "PIN_10"]
output_pins = ["PIN_12", "PIN_13", "PIN_14", "PIN_15", "PIN_16"]

[[split.peripheral]]
rows = 4
cols = 5
row_offset = 0
col_offset = 5
serial = [{ instance = "PIO0", tx_pin = "PIN_1", rx_pin = "PIN_1" }]
[split.peripheral.matrix]
matrix_type = "normal"
input_pins = ["PIN_7", "PIN_8", "PIN_9", "PIN_10"]
output_pins = ["PIN_16", "PIN_15", "PIN_14", "PIN_13", "PIN_12"]

```

jerkydev, 2025-07-06:
I tried
```
[rmk]
debounce_time = 30
```
and
```
[rmk]
debounce_time = 50
```
Unfortunately, this didn't help.

jerkydev, 2025-07-06:
Hmm, It's better now with 
```
debounce_time = 60
```
Isn't it too big? I think I now see a visual delay while typing. I googled that QMK uses 5ms debounce time by default and this issue isn't reproducible with QMK.

HaoboGu, 2025-07-06:
yeah, generally debounce_time is about 10-30ms. Interesting, is there only one key that repeats? or all keys occasionally repeat themselves

jerkydev, 2025-07-07:
It's not just one key also it's not all keys. Mostly affected keys are: k , . ;
I also think that I never noticed anything like this coming from the central (left) part of the keyboard.

HaoboGu, 2025-07-07:
@jerkydev Could you try #444?

HaoboGu, 2025-07-08:
nice

PR #444 (2025-07-08): Fix occasional repeat message from peripheral
Fixes #441
closed 2025-07-08 by HaoboGu