rmk-rs/rmk #481: Macros case error

Songggggr, 2025-07-17:
One uppercase S one lowercase s.
[image]

Actual output, All uppercase.
[image]

HaoboGu, 2025-07-17:
I cannot reproduce it, did you save the macro to the keyboard using the "save" button?

HaoboGu, 2025-07-17:
Interesting, do you switch capslock using shift? or the shift is used for switching input methods

Songggggr, 2025-07-17:
> Interesting, do you switch capslock using shift? or the shift is used for switching input methods

@HaoboGu Use rime, shift to switch Chinese and English

HaoboGu, 2025-07-17:
> [@HaoboGu](https://github.com/HaoboGu) Use Rime, shift to switch Chinese and English

Yeah, that's the reason. Actually macros do nothing magic -- it enables shift to print capitalized char, so ideally if you type very fast, same issue will happen. The only thing that the firmware can do is to increase the delay when switching the caps state. I opened a PR: #482.

lotem, 2025-07-17:
@HaoboGu Per online discussion, we can try inserting `LShift` key code "around" shifted characters.

HaoboGu, 2025-07-17:
[cornix-left.uf2.zip](https://github.com/user-attachments/files/21299784/cornix-left.uf2.zip)

@Songggggr Could you try this fi

HaoboGu, 2025-07-18:
Can you reproduce the input method switching on other keyboards? I think the firmware has nothing to do with this.

And as for the disconnection issue, yeah, you need to update both left and right firmware. Here it is:

[firmware.zip](https://github.com/user-attachments/files/21308640/firmware.zip)

Sorry for the inconvenience

Songggggr, 2025-07-19:
@HaoboGu There is no other keyboard that supports Vial, so you can't do the test.

Songggggr, 2025-07-19:
@HaoboGu Very perfect, no problem

Songggggr, 2025-07-19:
@HaoboGu There is only one problem that still bothers me, that is, the problem that macbook will automatically wake up.

Songggggr, 2025-07-19:
@HaoboGu What is the interval now? There are still a few apps with the same problem, but they are better than the previous firmware.

HaoboGu, 2025-07-19:
because you're using rime, you can ask @lotem about the details