rmk-rs/rmk #490: Proposal: Tap Hold Dance

HaoboGu, 2025-07-21:
## Definition

While implementing tap-dance(#479), I realized that there's a chance to merge tap-hold and tap-dance to a more powerful behavior -- it's call it tap-hold-dance（forgive me for the poor naming).

The tap-hold-dance key works like this:

- Tap a key multiple times to trigger different actions (just like tap-dance).
- After tapping, you can hold the key to trigger another action — the hold action changes based on how many times you tapped before.

The representation is actually same with current TapDance behavior:

```rust
struct TapHoldDance {
  tap_actions: [KeyAction; MAX_TAP_NUM],
  hold_actions: [KeyAction; MAX_TAP_NUM],
  tapping_term: Duration,
}
```

When the `MAX_TAP_NUM` = 1, it fallbacks to a normal tap-hold action, and when `hold_actions` is empty, it fallbacks to tap-dance action.

## Limitations

It sounds great, but it has some limitations still.

1. In Rust's type system, the `MAX_TAP_NUM` should be known at the compile time, and all `TapHoldDance` should have same length of `MAX_TAP_NUM`. That increases the memory usage A LOT, because if you have a tap-dance behavior with `MAX_TAP_NUM = 4`, then ALL normal tap-hold behavior requires 4x memory which will never been used.
2. Now tap-dance implementation accepts `KeyAction` as the behavior, that's why it's **u8 indexed**:
    ```rust
    enum KeyAction {
      ...
      TapDance(u8)
    }
    ```
    To avoid recursion we have to use the u8 as the index and find the corresponding `TapDance` from the `BehaviorConfig`. 

    Of course we can use `Action` in `tap_actions` and `hold_actions`, if we do that the `OneShot` keys(which is now a `KeyAction`) cannot be used in tap-hold-dance. It's lots work to do to migrate `OntShot` to `Action` too.
3. There are many options available to tap-hold, for example, `prior_idle_time`, `permissive_hold`, `chordal_hold`, etc. Those behaviors improves tap-hold keys. I'm not sure whether they can be kept easily for tap-hold-dance behavior.

tib888, 2025-07-28:
> Yes, it's impossible to save KeyActions for all possible combinations. For each morse key, we can build a action map, whose key is the operation sequence, like `0010` = `tap-tap-hold-tap`, and the value is the interpreted `KeyAction`.

The operation sequence could be stored in binary format, like you said, shifting in a '0' for taps, or a '1' for holds, but first you must shift in a '1' to mark the start of the sequence. This way no extra counter is needed but arbitrary length sequences could be described. In this format an u8 would be enough for a leading bit + maximum 7 taps/holds => means 1+2+4+8+16+32+64 = 127 possibilities, which "must be enough for everyone" (at least for the standard morse code table and even more)).

```
pub enum KeyState {
    /// After a press event is received.
    /// The number represents the number of completed "tap".
    Held(u8),
    /// Idle state after tap n times for morse keys
    IdleAfterTap(u8),
    /// Idle state after hold released
    IdleAfterHold(u8),
    /// The current key is a component of a combo, and it's waiting for other combo components
    WaitingCombo,
    /// Tap key has been processed and sent to HID, but not yet released.
    /// The number is used for morse keys, represents the number of completed "tap"s
    PostTap(u8),
    /// Key is being held, but not yet released
    /// The number is used for morse keys, represents the number of completed "tap"s
    PostHold(u8),
    /// Key needs to be released but is still in the queue,
    /// it should be cleaned up in the main loop regardless
    Release,
}
```

Each 'u8' in this enum could be encoded based on the description above.

Then after completition, in the morse behavior config the action table could be a simple list of (u8, KeyAction) of used pairs -> it would be pretty fast to search the maximum 127 items (I think 1..4 would be the typical). And would not use more memory than the current tap-dance configuration currently.

//morse codes example
(0b1_0u8, k!(E)) //tap "."
(0b1_1u8, k!(T)) //hold "-"
(0b1_00u8, k!(I)) //double tap ".."
(0b1_01u8, k!(A)) //tap hold ".-"
(0b1_1000u8, k!(B)) //hold tap tap tap "-..."

Not to mention that the tap dance, layer tap, modifier tap, tap hold and all other similar existing behaviors could be converted into this for uniform representation during the execution, and after that the excuting code could be simplified...

```
//tap dance codes example to enter language characters
tap_dances = [ { tap = "O", hold = "Equal", double_tap = "Kc0", hold_after_tap = "LeftBracket"} ] #o ó ö ő =>

(0b1_0u8, k!(O)) //tap: 'o' 
(0b1_1u8, k!(Equal)) //hold: 'ó' on HUN keyboard
(0b1_00u8, k!(Kc0)) //double tap: 'ö' on HUN keyboard
(0b1_01u8, k!(LeftBracket)) //tap hold: 'ő' on HUN keyboard
```

```
//modifier tap example on home row
or MT(S, LAlt) =>

(0b1_0u8, k!(S)) //tap
(0b1_1u8, k!(LAlt)) //hold
```

```
//modifier tap example on a thumb key
or LT(2, Space) =>

(0b1_0u8, k!(Space)) //tap
(0b1_1u8, mo!(2)) //hold
```

...

tib888, 2025-07-29:
I would say to save memory lets start with an u8, since it is very simple to extend to more if really needed.

Even 6 operations can encode a lot: 

[image]

[image]

tib888, 2025-07-31:
One idea for saving memory:

Tapdance using only an index:
```
    /// Tap dance action, references a tap dance configuration by index.
    TapDance(u8),
```

if morse would do the same:
```
    /// Morse action
    Morse(Morse<TAP_DANCE_MAX_TAP>),
```
should be replaced with this:
```
/// morse action, references a morse dance configuration by index.
Morse(u8),
```
+
```
/// Configuration for morse behavior
#[derive(Clone, Debug)]
pub struct MorseConfig {
    pub morse_action_sets: Vec<Morse<TAP_DANCE_MAX_TAP>, MORSE_MAX_NUM>,
}
```

tib888, 2025-07-31:
KeyAction is a sum type so its memory usage is about the same as its largest element, which is currently Morse.

With the above proposed Morse(Morse<TAP_DANCE_MAX_TAP>) => Morse(u8) reduces its memory usage from about 120 byte to 1 byte (+ the tag of the sum type)

After this Combo will not be a problem.

tib888, 2025-07-31:
+ a sidenote: 
If the structures a loosely packed, that can increase the memory usage too.
In C it can be controlled by #pragma pack(1), in rust, I have never tried similar, but likely #[repr(packed(1))] will do the same - it may worth a try if the above would not be enough, which I doubt.

tib888, 2025-08-01:
would it be possible to store this "expanded macro" struct only in the matrix of the "activated layer", while the configured layers of the keymap remain compact by using only the Morse(u8) form?

tib888, 2025-08-01:
Also, the combo input config should be based on key positions, even if it is not vial compatible.

HaoboGu, 2025-08-01:
I don't know. It seems not very easy, so I decided to keep it as is for now.

As for the positional combo, I want it very much! But I don't have enough bandwidth now, so maybe later

HaoboGu, 2025-08-02:
I'll have a look soon.

> Also I want to add a per key-position config matrix, in which home-row, left/right hand can be marked

I want it too! Per-key configuration is really important, in the future it should be an essential feature imo

HaoboGu, 2025-08-02:
> The main idea is that into_morse function is called at some key places, so the HeldKey.morse option will be filled, and can be used for further tap hold processing.

I found that the `TapHold(tap, hold)` is added back, this is where I stuck. I think in the future, both `TapHold` and `TapDance` should be removed, only `Morse` kept in the `KeyAction`. But to be compatible with vial, it's acceptable to have `TapHold` and `TapDance` as an alternative.

tib888, 2025-08-02:
actually all classic vial compatible actions could have stayed in the KeyAction, into_morse could convert them all, but until then they must remain 'recognizable' because for forks, non-positional combos those are the triggers (and also their output)....

tib888, 2025-08-02:
If there would be a wote, I would wote to get rid of vial support, or at least solve the clear_storage compile twice hack, when i want to upload the toml based config.

But lets return to the subject: **should i continue to work on this branch or not?**
Have you compared the memory usage?

HaoboGu, 2025-08-19:
Closed by #530