rmk-rs/rmk #496: Conflicting rolling key issues with permissive_hold

guicostaarantes, 2025-07-22:
Below are two scenarios that I am facing with my current layout using RMK. I'm replicating my current ZMK config and both scenarios work as expected for me on ZMK.

## Scenario 1 (only happens when `permissive_hold = true`)

I would like to type **"a+"**.

`KC_A` is on layer 0 (default), and `KC_KP_PLUS` is on layer 1. Activating layer 1 is done by holding `LT1(KC_SPACE)`.

I press `KC_A` > `LT1(KC_SPACE)` > `KC_KP_PLUS`

When I release `KC_A` before pressing `LT1(KC_SPACE)`, all goes well and I get **"a+"** at the end. However, if I press `LT1(KC_SPACE)` before releasing `KC_A`, I get the tap behavior even though I haven't released `LT1(KC_SPACE)`, ending up with **"a b"** (since the key for `KC_KP_PLUS` is a `KC_B` on layer 0).

## Scenario 2 (only happens when `permissive_hold = false`)

I would like to type **" b"**.

`KC_B` is on layer 0 (default). Space is sent by tapping on `LT1(KC_SPACE)`.

I press `LT1(KC_SPACE)` > `KC_B`

When I release `LT1(KC_SPACE)` before pressing `KC_B`, all goes well and I get **" b"** at the end. However, if I press `KC_B` before releasing `LT1(KC_SPACE)`, I end up with **"b "** instead of **" b"**. I believe this happens because `KC_B` is not a modkey so it computes on press, whereas `LT1(KC_SPACE)` computes the tap on release (it waits to check if it was a tap or a hold).

## Summary

I don't think Scenario 2 is a bug, since the behavior I want is the described at https://docs.qmk.fm/tap_hold#rolling-keys with PERMISSIVE_HOLD turned on, so it makes sense that it would not happen when I use `permissive_hold = false`.

Scenario 1 looks like a bug to me. The keypress of a non-modkey should not influence the tap-hold decision of a key pressed afterwards. If not a bug, it's at least something the user should be able to toggle on or off in their config. Please let me know if this already exists and I missed it.

Apart from that one issue, I'm able to daily drive a 36-key perfectly. VIA + mouse commands + wireless written in Rust, truly awesome.

HaoboGu, 2025-07-22:
thanks for the feedback, I will investigate it tomorrow to see if I can reproduce it. I'll report here if I found something

hitsmaxft, 2025-07-23:
Scenario 1: I'm sure it's bug that  releasing  key a  will lead to  a clear buffer behavior and make all following key into tapping behavior 

Scenario 2: since  PH is disabled,   two keys are processed individually, so  b is tap and lt is tap(output on releasing) laterly.

HaoboGu, 2025-07-23:
nice, I'm merging it