rmk-rs/rmk #551: Insufficient flash on stm32f072c8

clackups, 2025-09-14:
I attempted to port this keyboard to RMK:  https://www.aliexpress.us/item/3256807351467897.html

Here's my QMK config after reverse-engineering the board: https://github.com/clackups/qmk_userspace/tree/main/keyboards/clackups/xtips_x3s

I'm totally new to Rust and RMK, so probably there are ways to minimize the flash image size -- probably by stripping debug information or optimizing for size.  I did "rmkit init" and specified `stm32f072c8` as the MCU.  Cargo build produced the following errors:

```
  = note: rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 86800 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 116088 bytes
          rust-lld: error: section '.data' will not fit in region 'FLASH': overflowed by 116168 bytes
          rust-lld: error: section '.gnu.sgstubs' will not fit in region 'FLASH': overflowed by 116192 bytes
          
          flip-link: the native linker failed to link the program normally; please check your project configuration and linker scripts
          

error: could not compile `xtips_x3s` (bin "xtips_x3s") due to 1 previous error
```

clackups, 2025-09-14:
the tip for removing `defmt_rtt` also doesn't work

clackups, 2025-09-14:
after using the "Disable storage feature and defmt feature " instructions and unstable feature, it finally fits the flash size

HaoboGu, 2025-09-14:
> well, the chip has 128KB flash, and the binary image is 2537544 bytes, so no way to load it anyway...

128KB should be enough, 2537544 bytes is not the binary size, it's elf, you can use the following command to see the actual size:

```
cargo install cargo-binutils
cargo size --release
```

clackups, 2025-09-14:
well, it doesn't show up as a USB device after booting. But so far, I got the idea how to work with it, and will probably make it work.

HaoboGu, 2025-09-15:
If the flashing was succeed but it doesn't work, I guess it's the RAM limitation. stm32f072c8 has only 16 Kbytes RAM, which is very limited. You can try to tweak the following parameters in `keyboard.toml`:

```toml
[rmk]
event_channel_size = 2
report_channel_size = 2
vial_channel_size = 1
flash_channel_size = 1
macro_space_size = 8
tap_dance_max_num = 0
fork_max_num = 0
```

and use `cargo size --release` to see whether the `bss` field is smaller

HaoboGu, 2025-09-16:
Interesting, the bss takes only 6576 bytes, it *should* work. I guess maybe you need some extra tweaking for the usb peripheral, for example, rcc setting? Sorry I don't have a stm32f072 board. 

Do you have a debug probe? Things could be easier if you have logs.

HaoboGu, 2025-09-17:
I'll give it a try after my board arrives

clackups, 2025-09-17:
probably, something's wrong with my flashing procedure?

```
$ cargo +nightly build --release

$ ~/src/elf2dfuse/elf2dfuse target/thumbv7em-none-eabihf/release/xtips_x3s xtips_x3s.dfu

$ dfu-util -D xtips_x3s.dfu
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
Multiple alternate interfaces for DfuSe file
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #1 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Option Bytes  "
dfu-util: Non-valid multiplier ' ', assuming bytes
DfuSe interface name: "Internal Flash  "
File contains 1 DFU images
Parsing DFU image 1
No target name
Image for alternate setting 0, (2 elements, total size = 54988)
Setting Alternate Interface #0 ...
Parsing element 1, address = 0x08000000, size = 54948
Erase   	[=========================] 100%        54948 bytes
Erase    done.
Download	[=========================] 100%        54948 bytes
Download done.
Parsing element 2, address = 0x0800d6a8, size = 24
Erase   	[=========================] 100%           24 bytes
Erase    done.
Download	[=========================] 100%           24 bytes
Download done.
Done parsing DfuSe file
```

HaoboGu, 2025-09-17:
oh, you forgot to change the target: https://rmk.rs/docs/user_guide/2-2_local_compilation.html#_2-choose-your-hardware-and-install-the-target

closed 2025-10-23 by clackups